Xinyuan-LilyGO/T-Echo #2: LoRaWAN support

hkicko, 2021-09-06:
Could you please add an example using LoRaWAN? 
I found 2 libraries that support this radio - 
https://github.com/LacunaSpace/basicmac 
and
https://github.com/ngraziano/LMICPP-Arduino
BaicMac seems more promising. But if I try to use it, it fails the assert in this method in radio-ax126x.c:
```
// reset radio
static void radio_reset (void) {
    // drive RST pin low
    bool has_reset = hal_pin_rst(0);

    // If reset is not connected, just continue and hope for the best
    if (!has_reset)
        return;

    // wait > 100us
    hal_waitUntil(os_getTime() + ms2osticks(1));

    // configure RST pin floating
    hal_pin_rst(2);

    // wait 1ms?
    hal_waitUntil(os_getTime() + ms2osticks(1));

    // check reset value
    ASSERT( ReadReg(REG_LORASYNCWORDLSB) == 0x24 );

    // initialize state
    state.sleeping = 0;
}
```

FabTangi, 2021-09-10:
could you also confirm this device can be used on The Things Industries / Networks ?

hkicko, 2021-09-10:
Yes, that's the goal - to use it as a mapper for Helium or TTN.

LilyGO, 2021-09-13:
Hello, thank you for your proposal. We are increasing the support of loRaWAN for T-Echo. If you gain anything, you are welcome to contribute to T-Echo.

wbefield, 2022-01-14:
I have a T-echo and would like to use it on the Helium network Mapper

closed 2022-07-12 by liangyingy

MAY4VFX, 2023-12-03:
> https://github.com/Xinyuan-LilyGO/T-Echo/tree/main/examples/LoRaWAN
such a path does not exist, does T-echo TTN support any news?