Xinyuan-LilyGO/T-Echo #3: DeepSleep power consumption

FabTangi, 2021-09-10:
Can we have detailed infos on the power consumption of the product? (working / deepsleep)

LilyGO, 2021-09-11:
Working current: refer to https://github.com/lyusupov/SoftRF/wiki/Badge-Edition#battery-life
Sleep current: about 0.25mA

FabTangi, 2021-09-13:
Thanks for your answer. I was expecting a few microA, according to datasheets of components. Do you have plan to reduce the consumption in deep-sleep?

LilyGO, 2021-10-19:
Hi, There are no plans, and its sleep power consumption is already very low

closed 2021-10-19 by LilyGO

Llgok, 2024-09-09:
> Thanks for your answer. I was expecting a few microA, according to datasheets of components. Do you have plan to reduce the consumption in deep-sleep?

@FabTangi 
Hello, the latest low-power example has been updated. You can try using the example [Sleep\_Display](./examples/Sleep_Display) to reduce the power consumption of your onboard board.