Xinyuan-LilyGO/T-Echo #7: Drawing text on epaper display

CharlesMod, 2021-11-22:
Hello,

I've been trying for some time to successfully show text on the display, can you provide an example that shows how to do this?

LilyGO, 2021-11-22:
Here is a simple display  [example](https://github.com/Xinyuan-LilyGO/LilyGO-T-Echo/blob/main/examples/Display_lilygo/Display_lilygo.ino)

CharlesMod, 2021-11-22:
Thank you for your quick response!

I have already tried this example, trouble is it only shows bitmap print feature, not text print feature. Can you provide a different example?

CharlesMod, 2021-11-22:
FYI tested the following, just makes the code hang when included in the example you provided: 

`  display->setRotation(2);
  display->fillScreen(GxEPD_WHITE);
  display->setTextColor(GxEPD_BLACK);
  display->setFont(&FreeSansBold9pt7b);
  display->setCursor(0, 45);
  display->print("Hello World"); //crashes here
  display->update();`

LilyGO, 2021-11-22:
void setup()
{
    Serial.begin(115200);
    delay(200);
    boardInit();
    delay(200);
    LilyGo_logo();
/*************************************************/
delay(200);
display->setRotation(3);
display->fillScreen(GxEPD_WHITE);
display->setTextColor(GxEPD_BLACK);
display->setFont(&FreeMonoBold12pt7b);
display->setCursor(0, 45);
display->print("Hello World"); //crashes here
display->update();
/*************************************************/
}

CharlesMod, 2021-11-22:
Hm no dice, the //crashes here line when uncommented breaks the sketch, the RGB blinking stops

Here is the full code, does it work on a unit you have there?

```
/*
  Display lilygo logo example

  This example shows an e-Peper display.The RGB indicator blinks in cycles.

  This example code is in the public domain.
*/

#include "utilities.h"
#include <SPI.h>
#include <Wire.h>

#include <GxEPD.h>
//#include <GxGDEP015OC1/GxGDEP015OC1.h>    // 1.54" b/w
//#include <GxGDEH0154D67/GxGDEH0154D67.h>  // 1.54" b/w
#include <GxDEPG0150BN/GxDEPG0150BN.h>  // 1.54" b/w 

#include GxEPD_BitmapExamples
#include <Fonts/FreeMonoBold12pt7b.h>
#include <GxIO/GxIO_SPI/GxIO_SPI.h>
#include <GxIO/GxIO.h>

void setupDisplay();
void enableBacklight();
void configVDD(void);
void boardInit();
void LilyGo_logo(void);

SPIClass        *dispPort  = nullptr;
SPIClass        *rfPort    = nullptr;
GxIO_Class      *io        = nullptr;
GxEPD_Class     *display   = nullptr;

uint32_t        blinkMillis = 0;
uint8_t rgb = 0;

void setup()
{
  Serial.begin(115200);
  delay(200);
  boardInit();
  delay(200);
  LilyGo_logo();
  delay(200);
  display->setRotation(3);
  display->fillScreen(GxEPD_WHITE);
  display->setTextColor(GxEPD_BLACK);
  display->setFont(&FreeMonoBold12pt7b);
  display->setCursor(0, 45);
  display->print("Hello World"); //crashes here
  display->update();
}

void loop()
{

  if (millis() - blinkMillis > 1000) {

    blinkMillis = millis();
    switch (rgb) {
      case 0:
        digitalWrite(GreenLed_Pin, LOW);
        digitalWrite(RedLed_Pin, HIGH);
        digitalWrite(BlueLed_Pin, HIGH);
        break;
      case 1:
        digitalWrite(GreenLed_Pin, HIGH);
        digitalWrite(RedLed_Pin, LOW);
        digitalWrite(BlueLed_Pin, HIGH);
        break;
      case 2:
        digitalWrite(GreenLed_Pin, HIGH);
        digitalWrite(RedLed_Pin, HIGH);
        digitalWrite(BlueLed_Pin, LOW);
        break;
      default :
        break;
    }
    rgb++;
    rgb %= 3;
  }
}

void LilyGo_logo(void)
{

  display->fillScreen(GxEPD_WHITE);
  display->drawExampleBitmap(BitmapExample1, 0, 0, GxEPD_WIDTH, GxEPD_HEIGHT, GxEPD_BLACK);
  display->update();
}

void enableBacklight(bool en)
{
  digitalWrite(ePaper_Backlight, en);
}

void setupDisplay()
{
  dispPort = new SPIClass(
    /*SPIPORT*/NRF_SPIM2,
    /*MISO*/ ePaper_Miso,
    /*SCLK*/ePaper_Sclk,
    /*MOSI*/ePaper_Mosi);

  io = new GxIO_Class(
    *dispPort,
    /*CS*/ ePaper_Cs,
    /*DC*/ ePaper_Dc,
    /*RST*/ePaper_Rst);

  display = new GxEPD_Class(
    *io,
    /*RST*/ ePaper_Rst,
    /*BUSY*/ ePaper_Busy);

  dispPort->begin();
  display->init(/*115200*/);
  display->setRotation(2);
  display->fillScreen(GxEPD_WHITE);
  display->setTextColor(GxEPD_BLACK);
  display->setFont(&FreeMonoBold12pt7b);
}

void configVDD(void)
{
  // Configure UICR_REGOUT0 register only if it is set to default value.
  if ((NRF_UICR->REGOUT0 & UICR_REGOUT0_VOUT_Msk) ==
      (UICR_REGOUT0_VOUT_DEFAULT << UICR_REGOUT0_VOUT_Pos)) {
    NRF_NVMC->CONFIG = NVMC_CONFIG_WEN_Wen;
    while (NRF_NVMC->READY == NVMC_READY_READY_Busy) {}

    NRF_UICR->REGOUT0 = (NRF_UICR->REGOUT0 & ~((uint32_t)UICR_REGOUT0_VOUT_Msk)) |
                        (UICR_REGOUT0_VOUT_3V3 << UICR_REGOUT0_VOUT_Pos);

    NRF_NVMC->CONFIG = NVMC_CONFIG_WEN_Ren;
    while (NRF_NVMC->READY == NVMC_READY_READY_Busy) {}

    // System reset is needed to update UICR registers.
    NVIC_SystemReset();
  }
}

void boardInit()
{

  uint8_t rlst = 0;

#ifdef HIGH_VOLTAGE
  configVDD();
#endif

  SerialMon.begin(MONITOR_SPEED);
  // delay(5000);
  // while (!SerialMon);
  SerialMon.println("Start\n");

  uint32_t reset_reason;
  sd_power_reset_reason_get(&reset_reason);
  SerialMon.print("sd_power_reset_reason_get:");
  SerialMon.println(reset_reason, HEX);

  pinMode(Power_Enable_Pin, OUTPUT);
  digitalWrite(Power_Enable_Pin, HIGH);

  pinMode(ePaper_Backlight, OUTPUT);
  //enableBacklight(true); //ON backlight
  enableBacklight(false); //OFF  backlight

  pinMode(GreenLed_Pin, OUTPUT);
  pinMode(RedLed_Pin, OUTPUT);
  pinMode(BlueLed_Pin, OUTPUT);

  pinMode(UserButton_Pin, INPUT_PULLUP);
  pinMode(Touch_Pin, INPUT_PULLUP);

  int i = 10;
  while (i--) {
    digitalWrite(GreenLed_Pin, !digitalRead(GreenLed_Pin));
    digitalWrite(RedLed_Pin, !digitalRead(RedLed_Pin));
    digitalWrite(BlueLed_Pin, !digitalRead(BlueLed_Pin));
    delay(300);
  }
  digitalWrite(GreenLed_Pin, HIGH);
  digitalWrite(RedLed_Pin, HIGH);
  digitalWrite(BlueLed_Pin, HIGH);

  setupDisplay();
}
```

LilyGO, 2021-11-22:
I use your code. The normal operation. Are your factory firmware devices running properly?

CharlesMod, 2021-11-22:
I received 2 units in my order, one has a barely legible dim screen so I can't use that one, the other appears to work fine. If you send a compiled hex file, I could upload that to my hardware and check for functionality that way and get back to you

LilyGO, 2021-11-23:
Refer to [this ](https://github.com/Xinyuan-LilyGO/LilyGO-T-Echo/tree/main/examples/SoftRF_firmware) for uploading firmware

CharlesMod, 2021-11-24:
I've installed the firmware, the display on one appears to work just fine while the other is still very dim. I'm going to try to clean my installation directory for arduino and try the code I linked above again

CharlesMod, 2021-11-24:
Hm no dice, I cleared all of my libraries and reinstalled fresh from the lib folder but the device still says "LilyGo" rather than a blank screen with the words "hello world" shown as expected. 

[https://photos.app.goo.gl/YWjZ4oDC6KqnJfj6A](image of screen I'm seeing)

LilyGO, 2021-12-04:
One shows normal.One that's very dark?Are these two e-papers the same batch?

CharlesMod, 2021-12-06:
More like very dim than dark yes, they were included in the same shipment.

LilyGO, 2021-12-07:
Can you see the interface label of e-paper?  They are SYX......  ?

CharlesMod, 2021-12-07:
Opened up the two units to take a look, both boards are the same revision,
2021-6-28 (V1.0)

The displays are both SYX 2118 according to the screenprinting on the
ribbon cable. Other information is the same on both displays as well

HINK-E0154A07-A1
Date:2017-02-28
SYX 2118

CharlesMod, 2021-12-07:
Can you confirm the display driver library I should be using for this
particular screen?

Choices from the example sketches are:

#include <GxEPD.h>
//#include <GxGDEP015OC1/GxGDEP015OC1.h>    // 1.54" b/w
//#include <GxGDEH0154D67/GxGDEH0154D67.h>  // 1.54" b/w
#include <GxDEPG0150BN/GxDEPG0150BN.h>  // 1.54" b/w

LilyGO, 2021-12-10:
[image]
You chose the right driver

CharlesMod, 2021-12-10:
Hm okay, I still haven't been able to successfully draw text on these
screens unfortunately, do you have any other troubleshooting suggestions?

mkinney, 2022-02-04:
You could drop Meshtastic on the device(s) to see if the screen works. (grab the uf2 file in the firmware.zip file) https://github.com/meshtastic/Meshtastic-device/releases/tag/v1.2.53.19c1f9f

CharlesMod, 2022-02-04:
I've done so and the screen definitely works (though one of my units has a
screen that's super dim), but the issue is getting fresh code to compile
and push text to the screen successfully.

piotr-kubica, 2022-02-19:
Hi, I'm having the same issue. Seems it has something to do with AdafruitGfx lib
`
#include <SPI.h>
#include <GxEPD.h>

#include <GxGDE0213B72B/GxGDE0213B72B.h> // 2.13" b/w
#include <GxIO/GxIO_SPI/GxIO_SPI.h>
#include <GxIO/GxIO.h>

#define SPI_MOSI 23
#define SPI_MISO -1
#define SPI_CLK 18
#define ELINK_SS 5
#define ELINK_BUSY 4
#define ELINK_RESET 16
#define ELINK_DC 17

GxIO_Class io(SPI, ELINK_SS, ELINK_DC, ELINK_RESET);
GxEPD_Class display(io, ELINK_RESET, ELINK_BUSY);

#include <Fonts/FreeMonoBold9pt7b.h>

void setup() {
    Serial.begin(115200); 
    Serial.println("\n\n=== hello world ===");

    Serial.print("init_display...");
    SPI.begin(SPI_CLK, SPI_MISO, SPI_MOSI, ELINK_SS);

    display.init();
    Serial.println(" completed");

    delay(3000);
    display.setRotation(3);
    display.setTextWrap(false);
    display.fillScreen(GxEPD_WHITE);
    display.setTextColor(GxEPD_BLACK);

    display.setFont(&FreeMonoBold9pt7b);
    display.setCursor(20, 20);

// crashes here
// display.print("text") causes core panic exception !
//    display.print(String("text"));

    // this works ok
    display.fillRect(10, 10, 10, 10, GxEPD_BLACK);

    // this works ok
    display.drawChar(20, 40, 'a', GxEPD_BLACK, GxEPD_WHITE, 1);
    display.drawChar(30, 40, 'b', GxEPD_BLACK, GxEPD_WHITE, 2);
    display.drawChar(40, 40, 'c', GxEPD_BLACK, GxEPD_WHITE, 3);

    // also crashes here...
//    display.write('d');
    
    display.update();
}

void loop() {
    // put your main code here, to run repeatedly:
}
`

LilyGO, 2022-02-21:
Do you use t-Echo? This is NRF52840, not ESP32.

piotr-kubica, 2022-02-21:
@LilyGO do you have drivers for LILYGO®TTGO T5 V2.3.1 _ 2.13? Let me know I will try. Maybe it works.
This worked perfectly before I wonder what has changed.

LilyGO, 2022-02-22:
https://github.com/Xinyuan-LilyGO/LilyGo-T5-Epaper-Series

closed 2022-02-22 by LilyGO