Xinyuan-LilyGO/T-Echo #9: bootloader question

mkinney, 2022-02-04:
Thank you for providing the .hex file for the bootloader.

Can I create a uf2 using a tool like https://github.com/lancaster-university/codal/blob/master/utils/uf2conv.py and have it work okay on the T-Echo?

I ran this command:

```
% python3 uf2conv.py lilygo_techo_bootloader-0.6.1_s140_6.1.1.hex --convert --output techo.uf2
Converting to uf2, output size: 377856, start address: 0x0
Wrote 377856 bytes to techo.uf2
```

Ok to drop that uf2 file on to the device?

mkinney, 2022-02-04:
I tried copying that techo.uf2 file and it just copied the file to the device. So, no, it will not work.

Where can I get the uf2 bootloader that was used on my units?

```
contents of: INFO_UF2.TXT
UF2 Bootloader 0.6.1-2-g1224915 lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-293-gaf8e5a90) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: LilyGo T-Echo
Board-ID: nRF52840-TEcho-v1
SoftDevice: S140 version 6.1.1
Date: Oct 13 2021
```

LilyGO, 2022-02-09:
Hello, you need to double click the reset button.  UF2 file drag to BOOT:

mkinney, 2022-02-09:
I understand that is how you update **firmware**. I'm asking about the location of the **bootloader**.

Where is the source/download for the **bootloader**?

LilyGO, 2022-02-10:
[Reference](https://github.com/lyusupov/Adafruit_nRF52_Bootloader/tree/master/src/boards/lilygo_techo)

mkinney, 2022-02-10:
Thanks. Closing issue.

closed 2022-02-10 by mkinney