Xinyuan-LilyGO/T-Echo #10: Bluetooth examples

F4EWI, 2022-02-22:
Did you have an example of a bluetooth program for the T-ECHO ?

LilyGO, 2022-02-23:
Hello. [reference](https://github.com/Xinyuan-LilyGO/LilyGO-T-Echo#using-arduino-ide)，When you're done. Examples can be found under arduino IDE example
[image]
s

F4EWI, 2022-02-23:
The idea is to have a T-Echo communicate with a T-Watch 2020 via Bluetooth. I found only examples with BLE (one way). What would be the best way to get there? 
SerialToSerialBT on the T-Echo ?

LilyGO, 2022-02-24:
It has an example of BLE

F4EWI, 2022-02-25:
Which ?

LilyGO, 2022-02-26:
[image]

closed 2022-03-21 by LilyGO