Xinyuan-LilyGO/T-Echo #12: Programming a T-Echo with PlatformIO under Windows 10

F4EWI, 2022-03-16:
Do you have a tutorial that shows how to program a T-Echo under Windows 10 and Platformio environment?

LilyGO, 2022-03-17:
https://github.com/Xinyuan-LilyGO/LilyGO-T-Echo#use-platformio

F4EWI, 2022-03-17:
I need more than just a link. At first glance, some explanation is missing.
I may not have installed the correct version of J-Link??!!

[image]

LilyGO, 2022-03-21:
The J-link tool is required. As shown in figure https://github.com/Xinyuan-LilyGO/LilyGO-T-Echo/tree/main/bootloader#use-j-link-flash

F4EWI, 2022-03-21:
Thank you for the informations. Which model of SWD interface are you using?

LilyGO, 2022-05-12:
[reference](https://github.com/Xinyuan-LilyGO/LilyGO-T-Echo#use-platformio).  Article 5 Firmware can be downloaded via USB

closed 2022-05-12 by LilyGO