Xinyuan-LilyGO/T-Echo #16: Upload with platformio on Linux does not work

pjalocha, 2022-05-10:
The specified protocol jlink is not working, although the device is visible on the serial port /dev/ttyACM0

Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, nrfjprog, stlink
CURRENT: upload_protocol = jlink
Uploading .pio/build/nrf52840_dk_adafruit/firmware_signature.bin
SEGGER J-Link Commander V7.50a (Compiled Jul  8 2021 18:21:10)
DLL version V7.50a, compiled Jul  8 2021 18:20:53

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...FAILED: Cannot connect to J-Link.

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...FAILED: Failed to open DLL

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...FAILED: Failed to open DLL

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...FAILED: Failed to open DLL

Script processing completed.

LilyGO, 2022-05-11:
J-link is required

pjalocha, 2022-05-11:
Thus what exactly is required: I have T-Echo, connected to the USB port, it is recognized as /dev/ttyACM0, minicom works properly on this port, what else is needed to upload new code to the T-Echo ?

LilyGO, 2022-05-12:
J-link  download .  It's not a USB download. Arduino IDE can be downloaded via USB

LilyGO, 2022-05-12:
To change the configuration platformio.ini. 
upload_protocol = nrfutil  .
Because the boot program is different.  Before downloading, double click the reset button to connect t-Echo to DFU mode

closed 2022-05-31 by liangyingy