Xinyuan-LilyGO/T-Echo #17: Information on LEDs needed

surlyhacker, 2022-07-14:
I am really confused about the LEDs in this device.
Here are some questions:
1) how many total LEDs there are?
2) which ones are internal and which are external?  There is at least one internal blue LED that can be seen if you open the case or look through the hole on the back.  And clearly there are 2 external spots where LEDs can be seen.
3) What is the position of each LED in the device's case?
4) For each LED, is it single-color or are there any multi color LEDs?
5) What is the pin numbering for each LED?

Your existing pinout diagram shows this:
[image]

And that doesn't really answer most of my questions.

surlyhacker, 2022-07-14:
And if you are able to answer these questions, please also put the answers in widely accessible places and not just in this issue comments, such as:
1) On this github README
2) On your AliExpress listings
3) On your web site

GoldProto, 2022-08-06:
I am anticipating some answers on these issues as well. Especially curious as to why the blue LED seen inside the device runs non-stop when in sleep mode. A battery drain if you ask me... 
This needs to be corrected. 
Have you looked in the raw code to see if there are some lines to comment/uncomment to change the status of the internal LED's (not on the front)?

lewisxhe, 2023-04-14:
The blue light connected to GPS cannot be turned off. Unless the GPS is turned off, it cannot be turned off when the GPS is turned on.

lewisxhe, 2023-04-14:
GPS and Radio are controlled by P0.12, if this IO port is low level, then the GPS 1PPS light should not be on

surlyhacker, 2023-05-27:
I tried setting P0.12 to low during the Meshtastic shutdown code in my own fork of the firmware and it did not result in the 1PPS light turning off.  It was still on.  Why would this not work?

What about PWR_EN / P0.13 ?  Would that turn off the 1PPS LED?

lewisxhe, 2023-05-27:
1PPS cannot be closed.