Xinyuan-LilyGO/T-Echo #18: Possible bug with failing loading of meshtastic .UF2 files - BME 280 model only

GoldProto, 2022-07-30:
I have a number of the non-BME 280 T-Echo devices, all functioning when received. I have 1 of the BME 280 models that worked initially, but quit booting successfully immediately after updating the firmware. It sometimes partially boots for about 2-4 seconds (only the first boot after flashing new FW)- enough to activate a blue LED, refresh the screen and set the startup screen image, firmware #, etc.  But immediately a red light starts flashing (USB cable unplugged just before the glitch starts makes no difference) and most of the time starts repeatedly nonstop mounting/unmounting as a USB device on my laptop.
It does not work for any combination of firmware and bootloader hex/UF2 files I've tried. I've also tried compiling my own uf2 file from Meshtastic's main repo;  VSCode>PlatformIO>uf2conv.py>no success. 
What is strange is that is that the SoftRF firmware runs perfectly (sensors, comms, everything) along with all the demos in this (Xinyuan-LiliyGO/T-Echo) repository's example folder. 
Scratching my head. Would love to figure out what the bug is. 
Can someone help me resolve this issue?
Thank you.

lyusupov, 2022-08-06:
https://github.com/meshtastic/Meshtastic-device/issues/1565

GoldProto, 2022-08-06:
Update:
I tried your suggestion Lyusupov (thanks for making SoftRf by the way!) but it did not help. I now have 1 BME280 T-Echo device and 1 non-BME280 T-Echo device that are acting up.

- Both units still successfully load and run SoftRF - no problem!

- The non-BME unit I can get to run Meshtastic 1.2.64 - 1.3.31.0084643 **only one time** (after loading the uf2 file) until I put it to sleep. After trying to wake it up, it will not wake/start and run Meshtastic. I have to use a strange method to make it work this way, otherwise I cannot get it to run any version of Meshtastic whatsoever. This is the strange method I use to make this happen; 

1. Load [https://github.com/lyusupov/POST_TEcho/blob/main/bin/POST_T-Echo_CircuitPython6.uf2](url) to the non-BME280 device. (Device will not run the POST firmware)
2. Load T-echo bootloader (update-lilygo_techo_bootloader-0.6.1_nosd) 
3. Load a version of Meshtastic 1.2.64 to version  1.3.31.0084643. (newest)
4. Device will start successfully and run Meshtastic for one instance until put to sleep by holding button 2 down(lower left corner on device) and will display the proper "sleeping" screen with logo and Meshtastic version displayed.
5. Device will not wake/reset and start Meshtastic.

- I Cannot get the BME280 device to run any version of Meshtastic. It gets stuck in a weird bootloop mounting and unmounting as a usb serial device to the laptop. The only firmware I can get it to run successfully are all the examples in this folder except the Meshtastic firmware.  Link; [https://github.com/Xinyuan-LilyGO/T-Echo/tree/main/examples](url)

A very strange bug indeed!

LilyGO, 2022-08-09:
Hello，This [bootloader ](https://github.com/Xinyuan-LilyGO/T-Echo/blob/main/bootloader/lilygo_techo_bootloader-0.6.1-no-button-dfu-140_6.1.1.hex)  is required to use Meshtastic to wake up with button 2

GoldProto, 2022-08-09:
Thank you. I'll convert to .uf2 and see if it works.

GoldProto, 2022-08-10:
Thank you LilyGO. I used uf2conv.py tool to convert LilyGO's linked bootloader,  but the new bootloader (with new .uf2 firmware dropped in) did not fix the problem. The bootloader you suggested stopped the boot looping problem. 

One bad result is that I now cannot load and run the SoftRF program by @lyusupov .
The new INFO_UF2.TXT file has this text. The "SoftDevice: not found" probably is related to the new problem of not loading SoftRF.

`UF2 Bootloader 0.6.1-2-g1224915 lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-293-gaf8e5a90) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: LilyGo T-Echo
Board-ID: nRF52840-TEcho-v1
SoftDevice: not found
Date: Oct 13 2021
`

In the uf2conv.py script I used this number (bold) and perhaps it is causing an issue:

`python uf2conv.py -f **0xADA52840** -c -o lilygo_techo_bootloader-0.6.1-no-button-dfu-140_6.1.1.uf2 lilygo_techo_bootloader-0.6.1-no-button-dfu-140_6.1.1.hex
Converted to uf2, output size: 6955520, start address: 0x2000
Wrote 6955520 bytes to lilygo_techo_bootloader-0.6.1-no-button-dfu-140_6.1.1.uf2`

Any suggestions on a fix would be great.

LilyGO, 2022-08-11:
1. If you use SoftRF. Use this bootload [lilygo_techo_bootloader-0.6.1_s140_6.1.1.hex](https://github.com/Xinyuan-LilyGO/T-Echo/blob/main/bootloader/lilygo_techo_bootloader-0.6.1_s140_6.1.1.hex)

3.If you want to use Meshtastic. And want to use the button to wake up. Use this bootload[lilygo_techo_bootloader-0.6.1-no-button-dfu-140_6.1.1.hex](https://github.com/Xinyuan-LilyGO/T-Echo/blob/main/bootloader/lilygo_techo_bootloader-0.6.1-no-button-dfu-140_6.1.1.hex)

GoldProto, 2024-05-29:
I was able to recover my 2 failed devices a few days ago... finally after 2 years of having "softbricked" devices that could only run SoftRF formware correctly, it is great to have them now running meshtastic flawlessly!

This is what I did;

1. Installed the latest bootloader via Arduino IDE following these instructions;

Update Bootloader with DFU
To upgrade to the latest Bootloader + Softdevice using the serial port within Arduino IDE.
Select Tools > Board > Adafruit Feather nRF52840 Express
Select Tools > Programmer > Bootloader DFU for Bluefruit nRF52
Select Tools > Burn Bootloader
WAIT until the process complete ~30 seconds
Note: close the Serial Monitor before you click "Burn Bootloader". Afterwards, you shouldn't close the Arduino IDE, unplug the Feather, launch Serial Monitor etc ... to abort the process. There is a high chance it will brick your device! Do this with care and caution.

2. Performed a reset of the flash storage on the device - IMPORTANT STEP; YOU MUST have the Python environment installed along with the meshtastic CLI. The CLI command "meshtastic --noproto...then press any key" in the steps below is a critical step for this to work successfully!!  

To reset the flash storage on your nRF52 board:
Download and unzip the latest firmware from [Meshtastic Downloads](https://meshtastic.org/downloads).
Connect your device to your computer with a USB data cable.
Double click the reset button on your device (this will put it into bootloader mode)
Notice a new drive will be mounted on your computer (Windows, Mac, or Linux)
Open this drive and you should see three files: CURRENT.UF2, INDEX.HTM, and INFO_UF2.TXT
Find the file included in the downloaded firmware named Meshtastic_nRF52_factory_erase.uf2 and copy it onto the new drive. The device should reboot.
With the [Meshtastic CLI](https://meshtastic.org/docs/software/python/cli/) installed, open your Terminal/Console and enter the command: meshtastic --noproto.
Press any key, you should see the message: Formatting... Done.

3. Installed the latest .uf2 beta version of meshtastic here using drag and drop .uf2 firmware update method
(https://github.com/meshtastic/firmware/releases/tag/v2.3.10.d19607b)
(this version does not utilize the fuction of the touchsense button on the upper left corner of the t-echo device. Works great otherwise!)

@LilyGO , hope this solution can be passed on to other users that unintentionally softbrick their devices when trying a harmless .uf2 firmware upgrade!

closed 2024-05-31 by GoldProto