Xinyuan-LilyGO/T-Echo #29: How to disable the single blue (1PPS) LED

ajorg, 2023-06-26:
There is an explanation at https://github.com/meshtastic/firmware/issues/1493#issuecomment-1205024423 that seems to suggest that this blue LED cannot be disabled (not the blue LED from the tri-color LED, but the blue LED that's connected to the GPS 1PPS). But it's unclear if the meaning is that the LED cannot be disabled while the GPS is enabled, or if correctly disabling the GPS module should also disable the LED.

If disabling the GPS (or 1PPS) module correctly should disable its blue LED, clarification is needed as to the correct method of disabling the module.

If it's not possible, we'd appreciate positive confirmation that this blue LED cannot be disabled under any circumstance.

nxxxse, 2023-07-16:
I am also affected by this on my T-Echo with Mestastic 2.1.18. I continued where @richteel left of in https://github.com/meshtastic/firmware/issues/1493#issuecomment-1598008233. I checked with a Multimeter.

> Is the L76K really in standby?

Yes. https://github.com/meshtastic/firmware/blob/c75965480f9da9f067cf0d4722c06bf31ca7f959/src/gps/GPS.cpp#L629  works and pulls the WAKEUP pin on L76K to GND. I measured 0.0V. Meshtastic is off/in deep sleep and the internal blue LED is on. I measure 2.5 V on the 1PPS pin. I would have expected 3.3 V because that is VCC of L76K even in deep sleep of nrf52840.

RESET_N is 0 V … Interesting. Theory is that the GNSS module constantly resets and 1PPS is LOW in that state? I tested by putting:

```c
    digitalWrite(PIN_GPS_REINIT, 1);
    pinMode(PIN_GPS_REINIT, OUTPUT);
```

into `GPS::sleep()`. I confirmed that RESET_N is now 3.3 V with that patch in T-Echo deep sleep but still, the LED is still on.

What is interesting is that when T-Echo is on and I probe GND and RESET_N in voltage measuring mode, the internal blue LED comes on whenever I touch both pins. RESET_N seems quite sensitive and the high resistance of my multimeter is enough to trigger it. I measure 3.3 V so it is not like the ~1 MΩ of my multimeter would pull the pin down to GND. I don‘t have a oscilloscope here to look into this further.

> If it is, why is 1PPS held high?
> Is 1PPS really active low and should have the LED polarity flipped and connected to VCC rather than GND?
> Is it best to simply remove LED6? (Most likely this is the correct answer but I do not like to jump to conclusions. The previous questions need to be answered first.)

I did that now as a quick fix. I have spent too much time on this. Was a nice exercise though!

FabianMNoss, 2023-07-30:
@nxxxse I have a scope here, but had no time yet to go into the details of this bug yet. If you are interested, let's try together to sort this out?