Xinyuan-LilyGO/T-Echo #32: Intermittent GPS failure - solved

dc1rdb, 2023-10-02:
Just wanted to point out an assembly flaw that can lead to GPS not getting a fix at all or intermittently. I had that problem on my newly acquired unit and found out the following:
Looking at the board, in the upper left hand corner there is a stripe of conductive mesh tape connecting the SMA socket ground to the GPS chip shielding. On my unit, that mesh tape was also in contact with the passive components to the right of the GPS antenna receptacle on the board, effectively shortening them. There was a patch of isolating tape on top of the conductive mesh tape, which is just the wrong way around.
I placed the patch of isolating tape on the passive components between GPS chip and SMA socket and laid the conductive tape over it. The mesh tape still connects the SMA ground with the GPS chip shielding. Now the GPS performes flawlessly. The photo shows the area after the modification.

[image]

goedzo, 2024-09-09:
Do you by any chance have a before picture? I also seem to have issues in getting a GPS fix.

dc1rdb, 2024-09-10:
Sorry, I don't. 
Just make sure there is isolating tape below the center portion of the conductive tape.
In addition, keep in mind that your very first GPS fix can take quite a long time (like 15 minutes) and the unit should be placed outside with clear view of the sky.

goedzo, 2024-09-10:
Thanks! I actually did open up the device to check it and I saw there was already a small piece of transparent tape under the mesh. So I guess they updated this already? And yes I also indeed notice that the first fix takes quite some time.