Xinyuan-LilyGO/T-Echo #35: Connect to xcsoar

BrianJSavage, 2024-01-08:
I am trying to connect to xcsoar running on my phone.
I connect bluetooth and the device appears in the devices list but I only get No Data appearing.

I have checked all the NMEA options in the Tool setup.

Help!!

Cheers

brian

777CHIP, 2024-01-20:
I am having the exact same issue on both of my T-Echo’s … Softrf

would really like a response

lyusupov, 2024-02-03:
[image]

<br>
<br>
<br>

#

<br>
<br>
<br>

[image]

<br>
<br>
<br>

#

<br>
<br>
<br>

[image]

<br>
<br>
<br>

#

<br>
<br>
<br>

[image]

BrianJSavage, 2024-02-03:
Thanks for the reply.

I did all that, the only change was setting protocol to OGNTP.

Still doesn't work, and still getting No Data on the port.

Also, on my XCSOAR the device shows up as BLE Sensor, not BLE Port as shown above. Not sure if this makes a difference.
I also used channel A as shown, as shown above which disables the built in GPS and sensors.

Cheers

Brian

BrianJSavage, 2024-02-03:
I did unpair. reset everything and it didn't work when I replied before.
I just tried it again after your last comment, and it now works.
Thinking about it, the issue is that I thought I had to pair through Bluetooth, which I had to do to get SoftRF to work.
So it was still paired while I tried to attach to xcsoar.
Please confirm the solution is to not have it paired through Bluetooth, and it will connect.
This might be worth emphasising somewhere for other people, as to me it is counterintuitive.
Thanks for your help. Much appreciated.
Brian

lyusupov, 2024-02-03:
[image]

BrianJSavage, 2024-02-09:
Thanks for your help. All working

closed 2024-02-09 by BrianJSavage