Xinyuan-LilyGO/T-Echo #38: When will more be in stock ?

m1ctk, 2024-01-28:
As subject, thanks !

lyusupov, 2024-02-03:
https://www.lilygo.cc/products/t-echo-nrf52840

#

[image]

m1ctk, 2024-02-03:
Thanks yes I noticed today, just ordered one

lyusupov, 2024-02-03:
[image]

Danie10, 2024-02-22:
I see the Ship By Date, but no option for Add to Cart. It just shows Sold Out in that area.

lyusupov, 2024-03-08:
[image]

<br>
<br>

1) https://www.lilygo.cc/products/t-echo
2) https://www.lilygo.cc/products/t-echo-nrf52840