Xinyuan-LilyGO/T-Echo #42: internal Flash Memory example??

richonguzman, 2024-05-03:
cant find any internal Flash Memory example to use this

lyusupov, 2024-05-10:
https://github.com/adafruit/Adafruit_SPIFlash

richonguzman, 2024-05-10:
> https://github.com/adafruit/Adafruit_SPIFlash

So I would need to read/write flash with flash examples or SDFlash examples?

PS: can flash memory be written with VSCODE>Platformio?

lyusupov, 2024-05-10:
> So I would need to read/write flash with flash examples ...

Access to T-Echo's 2 Mbytes QSPI flash memory is possible with FAT filesystem and USB Mass Storage method. See this example:

https://github.com/lyusupov/SoftRF/wiki/Badge-Edition.-Aircrafts-database

richonguzman, 2024-05-10:
so the flash data is added just the uf2 files when starting the USB Mass Storage method ??

on the same folder and all?

lyusupov, 2024-05-10:
1) the uif2 firmware binary is applicable when the device in Adafruit **bootloader mode** only
2) in application mode (Arduino) -it is possible to use the same 'Drag and Drop' method as well, read 
   - Adafruit SPIflash examples and
   - Adafruit TinyUSB Arduino examples

richonguzman, 2024-05-10:
> 1. the uif2 firmware binary is applicable when the device in Adafruit **bootloader mode** only
> 2. in **application mode** (Arduino) -it is possible to use the same 'Drag and Drop' method as well, [...]

sorry I got lost for a moment:

My The T-echo is running as "Adafruit bootloader mode" and I can upload like a "external mass flash disk" the .uf2 file and also through VSCODE+Platformio. This works fine.

But this is where I need to be specific: will I be able to add also a .json file into the flash of T-echo by copying it also to the "external mass flash disk" that "Adafruit bootloader mode" enables?

lyusupov, 2024-05-10:
> will I be able to add also a .json file into the flash of T-echo by copying it also to the "external mass flash disk ...

Play with **msc_external_flash** sketch _(mentioned above)_ to answer your question by yourself.
You may borrow T-Echo specific GPIOs and flash types from SoftRF source code:
https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/platform/nRF52.cpp#L199-L233

richonguzman, 2024-05-10:
> Play with **msc_external_flash** sketch _(mentioned above)_ to answer your question by yourself. You may borrow T-Echo specific GPIOs and flash types from SoftRF source code: [...]

thanks for the guidance!!!

 I will try (as this is all new to me) 

 ;)

lyusupov, 2024-05-10:
[image]

closed 2024-05-10 by richonguzman