Xinyuan-LilyGO/T-Echo #43: Tactile switch/screen connector part numbers

vitaminmoo, 2024-05-05:
Do you have the edge mount tactile switch part numbers available? I seem to have exploded mine and am not having luck finding a compatible part number for replacement.

Same with the board mount connector for the display - that came apart when I removed the battery (the adhesive tore it off)

vitaminmoo, 2024-05-05:
XF2M-2415-1A appears to be compatible with the ribbon cable connector

vitaminmoo, 2024-05-05:
P16763CT-ND looks maaaybe close enough to work on the tactile buttons

vitaminmoo, 2024-05-18:
Both are wrong - The switches are a doable fit, but the pads of each switch are grounded so it specifically needs to be a switch that bridges both the back contacts when pressed, not front to back or otherwise.

The ribbon cable connector is also wrong in two ways - it's not quite a fit length-wise for the mechanical pads, and I also miscounted so it's got two to many contacts

vitaminmoo, 2024-05-18:
FH12A-24S-0.5SH(55) actually looks really good for the fpc

smiba, 2024-07-02:
I'm having the same issue, I dropped my T-ECHO and one of the switches has destroyed itself upon impact.

Have you found a replacement part ?

vitaminmoo, 2024-07-08:
@smiba I unfortunately have not

vitaminmoo, 2024-07-09:
They may be 1TS003B-1400-3500A-CT - I'll update when I get my hands on some

joe-albanese, 2025-02-22:
I purchased the buttons @vitaminmoo mentioned and they work perfectly. They feel maybe a touch mushier, but that might just be because I soldered them on slightly crooked