Xinyuan-LilyGO/T-Echo #46: Screen not responding

josephmarino1, 2024-06-09:
Hi there,

Not sure if this is the right place to post. I recently unboxed a t-echo that I purchased a few months back. Since opening it, the screen has been stuck on the "Normal Off" screen. I have been able to update it to run Meshtastic 2.3.10.d19607b Beta, but I cannot pair it since the screen is unresponsive. 

I have tried everything I could find online as far as wiping the device via meshtastic's file, re-flashing the firmware to different versions, and flashing the bootloader using Arduino IDE software, but the screen never changes. 

Thanks!

Hal102, 2024-07-23:
Hi, 
I am in the same situation, I received a new T-Echo yesterday and on powering up the screen has been unresponsive. I have managed to conduct an update to the latest Meshtastic release but have not been able to make the screen work.

danielemery, 2024-12-22:
I am experiencing the same issue, the e-ink display no longer refreshes. I have copied over the latest firmware using meshtastic flasher, but still no refreshed e-ink display so I am unsure of the state of the device. I cannot connect to it with bluetooth.

mbertazz, 2025-02-04:
Same issue here, screen is unresponsive and cannot connect via Bluetooth.