Xinyuan-LilyGO/T-Echo #47: T-Echo nonreactive with red LED pulsing

stefanthoss, 2024-07-27:
My T-Echo with Meshtastic 2.3.15 was running fine but in the middle of the night it just froze. No more Bluetooth connection and the device doesn't react to any button presses. The display just shows whatever it showed when it froze.

The left of the two red LEDs makes two pulses every 5 seconds and that's it. I tried the following without success:
* Flashing the latest stable version of Meshtastic. The device can be brought into DFU mode with a couple click of the top button and it accepts the firmware via file copy. But it doesn't change the behavior of being completely nonreactive.
* Trying to [update the bootloader](https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/update-techo-bootloader) doesn't work either, the device accepts the `.uf2` file, disconnects, and nothing happens.
* Disconnecting the battery for a couple of hours, reconnecting it, flashing the bootloader, and flashing Meshtastic.

Any idea how I can rescue the device?

stefanthoss, 2024-08-01:
Rescued the device with the following steps (only this exact order of steps seems to work):
1. Updating the [bootloader using adafruit-nrfutil](https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/update-techo-bootloader/#method-2-using-adafruit-nrfutil) with v0.6.1 (the UF2 file method doesn't seem to work for me).
2. Erasing the flash of the T-Echo using https://flasher.meshtastic.org.
3. Flashing the new firmware using https://flasher.meshtastic.org.

But then 3 days later, the same thing happens again and the T-Echo becomes nonreactive. Do I have a defective device or are these known issues?

stefanthoss, 2024-08-06:
Same with the latest stable Meshtastic firmware v2.4.0.

wesker-albert, 2024-09-03:
Same situation with the two T-Echo devices I bought straight from Lilygo. Erased, reflashed. Works fine for maybe a few days, then corrupts itself again.

They're pretty much $50 paperweights currently. One was intended to be a mobile solution, but I can't even rely on it enough to take it with me.

InsigneousLeptogre, 2024-10-20:
I've had mine a few weeks and havent had this issue.

KptMorgen, 2024-12-16:
I have the same - bought 2 pc, one works just fine, the other froze just the way you describe. It seems to me from the description it's just the firmware though?

stefanthoss, 2024-12-26:
Closing this ticket. This was a software issue with the public LongFast channel on nRF52 devices, the issue did not occur on my T-Echo when I didn't have the default public channel configured. It seems to be solved in the newer 2.5.10+ versions as I haven't seen it in a while even with the default public channel. I read somewhere on Discord that there was a memory leak in earlier versions which caused the bug when there was a lot of nodes in the local node DB.

closed 2024-12-26 by stefanthoss