Xinyuan-LilyGO/T-Echo #57: I seem to have bricked my T-Echo

ketan, 2025-04-25:
Hi,

I seem to have bricked my t-echo. It no longer shows up as a USB drive when double-clicking the reset button. I've tried removing the battery etc. Is there a way to re-upload the bootloader and reflash?

I use macos sequoia (m4), if that makes any difference.

Thanks

lewisxhe, 2025-07-14:
Can you tell me what you did?
As long as no other program is written through the SWD interface, the boot is always there.

Vaanhool, 2025-07-24:
I have the same problem with two T-Echo devices. They are unusable for almost one year. If I remember correctly I uploaded new SoftRF firmware via DFU mode then all of them bricked. Doesn't respond to anything, no DFU mode. The red LED is constantly on, the blue LED is flashes once when I plug the USB C cable then a pause of about 12 seconds and after thet is it flashes a short every second.
It wasn't the first update, I hadn't experienced anything like this before.

I tried burn bootloader with Arduino IDE, but the board not responding or not in DFU mode and failure.

lewisxhe, 2025-07-25:
Double-click the upper left button. If you cannot enter DFU mode, there may be two reasons:
1. The motherboard is dead
2. Burning other firmware has damaged the boot program

You can only re-burn the boot firmware to see if the boot program can be written into t-echo normally.

The swd interface of t-echo is next to the usb. You need a jlink or daplink

[image]

ketan, 2025-07-25:
Thanks for the pinout. I'll take a stab at re-programming this.

Vaanhool, 2025-08-15:
I have had success with both T-echoes. I followed this tutorial:

[https://github.com/Xinyuan-LilyGO/T-Echo/blob/main/bootloader/README.MD](url)

Some advices:
- disconnect the batteries before flashing.
- you need 4 dupont female wire or this if you don't want remove the board from the case to solder a header for connection [https://www.aliexpress.com/item/1005009672504212.html](url)
- power the device with USB C cable
- erase the flash first, after that burn the bootloader

I ordered many SWD interface before that but only this one worked with the J-Flash (Lite).
[https://www.aliexpress.com/item/1005007350851129.html](url)

My problem was that I saved the SoftRf firmware with a right click and instead of binary code it was saved with html content. I overlooked this and copied it to the device, after that the DFU mode disappeared.

Thanks

Vaanhool, 2025-08-21:
> [@Vaanhool](https://github.com/Vaanhool) - what firmware.hex file did you flash — the bootloader, or softrf itself?

Hi,
I used the 0.6.1 adatfruit bootloader from this:
[https://github.com/lyusupov/SoftRF/tree/master/software/firmware/binaries/nRF52840/Adafruit_nRF52_Bootloader](url)

After that, you can put the device to DFU mode and copy the proper uf2 file to the UDisk.