Xinyuan-LilyGO/T-Echo #60: Damaged board with overvoltage, repair attempt

rmu75, 2025-08-12:
Faulty USB brick and/or moisture in the enclosure somehow damaged my T-Echo. The LDO, the nrf SOC and some FETs near the USB connector let the smoke out.

I found the schematics, got replacement parts and fixed things that looked broken (Q6, Q7, Q8, U11, also replaced U9 and D7 just to be sure) and somewhat successfully even replaced the SOC.

Now the 3v3 rail is stable, board draws about 140mA on 5V, and I could flash the bootloader via SWD while being powered via USB-C. But now I'm stuck -- the board won't enumerate on USB-C, and it doesn't respond anymore on SWD while powered via USB-C.

Powering via SWD with 3,3V, i can connect with openocd, halt target, look around, investigate flash etc..., all seems normal. Running the SOC from openocd causes pulsing red LED. 

Investigating a bit further, something seems wrong with the power supply VDD_nRF, powering via USB, that rail only goes up to 1,8V. According to schematics, this should be connected to VDD3V3 (which measures OK at 3,3V and seems to be stable when looking on the scope) via L8. Because of incomplete silksreen, I'm not sure where to look exactly for L8, continuity-testing only got me to some unpopulated footprint next to the crystal cluster near the LORA module.

So I'm stuck. Any help or hint would be greatly appreciated -- maybe the PCB artwork is available somewhere or at least identification which parts are placed where and if there are parts that are not populated. Thanks.

lewisxhe, 2025-08-13:
[image]

[image]

This should help you

rmu75, 2025-08-13:
> [T-Echo.pdf](https://github.com/user-attachments/files/21745700/T-Echo.pdf)
> 
> This should help you

Thank you very much, exactly what I was looking for, that is indeed very helpful.

closed 2025-08-13 by rmu75